micasense/imageprocessing #104: Capture Alignment Issue

and-viceversa, 2020-01-27:
First off, thank you for this library. It's super helpful from both an educational and production standpoint.

I successfully converted raw Captures to undistorted reflectance using both the tutorial and the capture.compute_undistorted_reflectance() method in my own code. The logical next step is to align the Images in my Captures so I can feed them into a photogrammetry suite.

I successfully used the Rig Relatives alignment tutorial to export a .png, but I'm having trouble using the capture.create_aligned_capture() method in my own code. I plot the undistorted reflectance images successfully, however when I use the create_aligned_capture() method, the output .tif stack has very heavy uniform vertical striping. The image also has an oblique look, even though the imagery was taken at nadir. I got these same vertical striping results using Rig Relative and Active Image Alignment.

I can add a code sample, but I haven't done anything outside the tutorial or basic Capture Class usage. Images were captured using a MicaSense RedEge-M firmware v6.0.1.

# correct undistorted reflectance
[image]

# correct rig relative tutorial png
[image]

# problem export from create_aligned_capture(). note the oblique look.
[image]

poynting, 2020-01-27:
Yes, please provide the simplest possible code example that creates the problem described, otherwise it's not possible to help.

poynting, 2020-01-27:
Also, please take a look at the batch processing code.  Outputting images into photogrammetry is the purpose.  I generally suggest running the Active alignment tutorial first, and then copying the warp_matrices it outputs at the bottom into the middle of the batch processing notebook.

and-viceversa, 2020-01-27:
Tested again with the Passive Rig Relative tutorial and Active Alignment tutorial, both save the aligned RGB and CIR as expected. 

Code snippet from Batch Tutorial using the same warp_matrices as successful passive and active alignment runs:
`# batch processing tutorial

# using panel_irradiance from the panel capture
# using same warp_matrices from the successful Active Alignment tutorial
alignment_capture.create_aligned_capture(irradiance_list=panel_irradiance, warp_matrices=warp_matrices, normalize=False,
                                         img_type=img_type)
alignment_capture.save_capture_as_stack(os.path.join(output_dir, 'image_stack.tif'))
alignment_capture.clear_image_data()`

Oddly enough with the tutorial code, I get the same multiprocessing error on the ImageUtils.align_capture() call from Issue #51 . My script appears to be importing itself in an infinite loop. I'm familiar with the fork/spawn problems from other projects so I'll take a look at this later. Multithreaded=False fixes it still.

poynting, 2020-01-27:
If the code is running the code in a separate script (outside of the notebook), I'm guessing it's necessary to condition execution with 
```
if __name__ == '__main__':
  #your code
```
to prevent recursion.  I just tested this by exporting the Alignment notebook as a `.py`, removing the ipython specific calls, and running it with and without the code conditioned in that way.  Also note that when the images showed, they needed to be closed before the script would continue.

and-viceversa, 2020-01-27:
Forgot to wrap with the `if ... '__main__'` block... good catch.

To add to the snippet above, I tried the `alignment_capture.create_aligned_capture()` with default `warp_matrices=None`. Still no joy. `capture.save_capture_as_rgb()` works as intended. I'm not sure where else to check since both tutorials work correctly, and the RGB output is correct too.

Trying a different image file yielded same results.

and-viceversa, 2020-01-27:
The initial issue above reflected running the imageprocessing library from my own environment.

I replicated the same issue following all macOS installation instructions, using the miniconda environment, and running the notebook code without modification. All pytests passed successfully.

The below image is the output from Alignment/Stack Export block. The RGB and CIR images saved correctly.

[image]

poynting, 2020-01-28:
Unfortunately I still haven't seen a complete code example, so it's difficult and time consuming for me to provide much assistance. Further, the images provided above show up to me as a jpeg, implying it's been read by some image viewer and parsed into that format.  The TIFF images are 6-band tiff images and won't be displayed properly by most normal TIFF file viewers.

Can you post a complete code example?
Can you post the actual output tiff? 
What software is used to open the the images that produces this result? 
Does it work if you open the image in QGis?

poynting, 2020-01-28:
When I save the following as `align_capture.py` and run `python align_capture.py` in the imageprocessing checkout directory I get the expected results.  3 `rgb_*.jpg` and 3 `stack_*.tif` files are created  and all of them appear correctly in their respective viewers (windows photo viewer for the jpg and QGis for the TIFF).  Windows photo viewer and windows previewer don't show the tif files correctly, but I wouldn't expect them to properly display 6-band data. 

[image]

[image]

```
#!/usr/bin/env python
# coding: utf-8

import micasense.imageset as imageset
import micasense.capture as capture
import os, glob
from numpy import array, float32

# Use the warp_matrices derived from the Alignment Tutorial for this Altum
warp_matrices=[array([[ 1.0044947e+00, -4.3070558e-03, -9.3133707e+00],
   [ 2.7266450e-03,  1.0027785e+00,  4.1896835e+01],
   [ 8.4233778e-07, -2.2866227e-06,  1.0000000e+00]], dtype=float32), array([[1., 0., 0.],
   [0., 1., 0.],
   [0., 0., 1.]], dtype=float32), array([[ 9.9804604e-01, -4.8819836e-03, -4.5177472e-01],
   [ 4.5076986e-03,  9.9605155e-01,  3.2124016e+01],
   [ 1.2246041e-06, -1.2118925e-06,  1.0000000e+00]], dtype=float32), array([[ 1.00085151e+00, -1.85801671e-03, -4.56634855e+00],
   [ 1.85797014e-03,  1.00130415e+00,  4.88208351e+01],
   [ 5.50759012e-07, -1.02554075e-07,  1.00000000e+00]], dtype=float32), array([[ 1.0023643e+00, -2.9937453e-03, -8.9271030e+00],
   [ 3.1016041e-03,  1.0009993e+00,  1.4952276e+01],
   [ 1.4310590e-06, -8.1405415e-07,  1.0000000e+00]], dtype=float32), array([[ 6.35209892e-02,  1.17877689e-05,  1.40322785e+01],
   [-4.56733969e-04,  6.35520044e-02,  1.15592432e+01],
   [-4.15804231e-06, -2.63551964e-06,  1.00000000e+00]])]

if __name__ == '__main__':

   imagePath = os.path.expanduser(os.path.join('.','data','ALTUM1SET'))
   panelNames = glob.glob(os.path.join(imagePath,'000','IMG_0000_*.tif'))
   imageNames = glob.glob(os.path.join(imagePath,'000','IMG_0245_*.tif'))
   print("Loading Panel capture")
   panelCap = capture.Capture.from_filelist(panelNames)
   panel_irradiance = panelCap.panel_irradiance(panelCap.panel_albedo())
   img_type='reflectance'

   print("Loading Flight capture")
   flightCapture = capture.Capture.from_filelist(imageNames)
   print("Captures loaded")

   print("Creating aligned stacks using warp_matrices defined above and panel irradiance")
   irradiance = panel_irradiance+[0]
   # Align images and save each capture to a layered tiff file
   flightCapture.create_aligned_capture(irradiance_list=irradiance, warp_matrices=warp_matrices)
   flightCapture.save_capture_as_stack('stack.tif')
   flightCapture.save_capture_as_rgb('rgb.jpg')
   flightCapture.clear_image_data()

   print("Creating aligned stacks using rig relatives only and panel irradiance")
   flightCapture.create_aligned_capture(irradiance_list=irradiance, warp_matrices=None)
   flightCapture.save_capture_as_stack('stack_relatives.tif')
   flightCapture.save_capture_as_rgb('rgb_relatives.jpg')
   flightCapture.clear_image_data()

   print("Creating aligned stacks using rig relatives and dls data only")
   flightCapture.create_aligned_capture(irradiance_list=None, warp_matrices=None)
   flightCapture.save_capture_as_stack('stack_relatives_dls.tif')
   flightCapture.save_capture_as_rgb('rgb_relatives_dls.jpg')
   flightCapture.clear_image_data()
```

and-viceversa, 2020-01-28:
Well, you burned me on this one.

You're right, macOS Preview can't display multiband tifs :/ . What really threw me was the odd zoomed in/oblique look, and I skipped over the obvious.

My flight images and the training data look normal in QGIS from both the tutorial and my own environment. Thanks again for your help and the awesome library from MicaSense. Consider this one closed.

poynting, 2020-01-28:
Glad that we got to the bottom of it.  There's a note in the Alignment tutorial about opening these stacks with a GIS, but it may be that we need to be more explicit that normal image viewers aren't likely to work. 

https://micasense.github.io/imageprocessing/Alignment.html#Stack-Export

closed 2020-01-30 by poynting