micasense/imageprocessing #107: MicaSense Image Processing Setup Error: "Provided file path does not have a band name"

aloboa, 2020-03-09:
I follow MicaSense Image Processing Setup, run and pass the 1st test ("Testing Installation"), 
```
Successfully imported all required libraries.
Successfully executed exiftool.
```
but get an error for the 2nd one ("Testing image reading and panel detection"). What can be wrong?

```
ValueError                                Traceback (most recent call last)
<ipython-input-4-22d275e27f9f> in <module>
      3 imageName = glob.glob(os.path.join(imagePath,'IMG_0000_1.tif'))[0]
      4 
----> 5 img = Image(imageName)
      6 img.plot_raw(figsize=(8.73,8.73));
      7 

~/micasense/imageprocessing/micasense/image.py in __init__(self, image_path, exiftool_obj)
     70 
     71         if self.meta.band_name() is None:
---> 72             raise ValueError("Provided file path does not have a band name: {}".format(image_path))
     73         if self.meta.band_name().upper() != 'LWIR' and not self.meta.supports_radiometric_calibration():
     74             raise ValueError('Library requires images taken with RedEdge-(3/M/MX) camera firmware v2.1.0 or later. ' +

ValueError: Provided file path does not have a band name: ./data/0000SET/000/IMG_0000_1.tif
```

aloboa, 2020-03-09:
Actually, pytest run produces errors:
`pytest . > pytestout.txt`

[pytestout.txt](https://github.com/micasense/imageprocessing/files/4307725/pytestout.txt)

poynting, 2020-03-09:
Did you install git-lfs before cloning the repository? See e.g. 
https://github.com/micasense/imageprocessing#how-do-i-get-set-up
https://github.com/micasense/imageprocessing/issues/100#issuecomment-572688850 
https://github.com/micasense/imageprocessing/issues/68#issuecomment-524124873

aloboa, 2020-03-09:
No, I did not have git-lfs 
(installing this package should be noted in 
https://micasense.github.io/imageprocessing/MicaSense%20Image%20Processing%20Setup.html
right after sudo apt install git)

I've installed it now, removed the micasense conda environment and directory and performed a fresh installation.
pytest still produces the same errors

It's weird because
`conda env create -f micasense_conda_env.yml`
does not report any problem

poynting, 2020-03-10:
It's likely necessary to wipe out the git cloned directory and re-clone the repository (not the conda environment, you can leave that alone, it's probably just fine). I haven't had luck with some of the workarounds if git-lfs was installed after the `git clone`.

The problem isn't with the python (conda) environment, it's with the data files that the tests use which are stored in git-lfs. If git-lfs is installed after the clone, the data files don't get downloaded during the clone so you only have git-lfs file pointers. There is no image data or metadata in the file pointers, so the tests can't pass.  I suspect there's a git lfs command to fix this, but I found it quicker to just re-clone the repository.

aloboa, 2020-03-10:
Solved by making a fresh installation with git-lfs on another machine.
Thanks

aloboa, 2020-03-10:
Surprisingly, do not get it to work on the previous machine.
This is what I do:

```
conda deactivate
conda remove --name micasense --all
sudo rm -r /home/npratf/micasense/imageprocessing
sudo rm -r /home/npratf/micasense
sudo apt remove git
sudo apt install git-lfs
mkdir micasense2
cd /home/npratf/micasense2
git clone https://github.com/micasense/imageprocessing.git
cd imageprocessing
conda env create -f micasense_conda_env.yml
conda activate micasense
pytest .
```

and still get the many errors
Provided file path does not have a band name:

TheFloe1995, 2020-03-25:
Had the same issue. Solved it with:
`git lfs install`
`git lfs pull`

closed 2020-07-02 by poynting

Hisschao, 2020-11-27:
The same problem has been encountered on the windows system, "ValueError: Provided file path does not have a band name: ./data/0000SET/000/IMG_0000_1.tif”，Have you solved the problem？

aloboa, 2020-11-27:
Frankly, do not quite remember as I finally could not accomplish what I wanted to do and gave up.
I think I was able to solve this particular issue in one machine but not in another (see above), linux in both cases.

poynting, 2020-11-27:
Git LFS was likely not installed prior to cloning the repository.

LBerger17, 2021-04-17:
Hi, what about trying to use this script with my own images collected with my altum camera? I was able to get it to work with the tutorial but I get the band name error trying to use my own data. Any advice?

julixax, 2021-09-16:
I have gone through all the trouble shooting options for resolving this issue.  I am running on Mac OS and I made sure to follow all the setup steps in order that was given in the set-up tutorial. I install gt-lfs before cloning but I get 74 fails and 44 errors from pytest. I then tried uninstalling and reinstalling all the set-up steps and I am running into the same errors and then get error from the first tutorial of "provided file path does not have a band name"

poynting, 2021-09-16:
@julixax can you verify that the files in the data directory are there and have a reasonable size? Can you run exiftool ok one of them and get metadata printed?  If not, it's likely a git lfs error.

julixax, 2021-09-16:
Yes I fixed the error by running 
git-lfs pull
only after the git clone command, not before. It seems to be working now

poynting, 2021-09-16:
Awesome