micasense/imageprocessing #109: stacked geotiffs: Adding Filename to a nested list, aligning

swawa, 2020-04-15:
Hi Poynting,
few questions:
(I am using an Altum, firmware version 1.3.6)

- a) is it possible, to add the filename to the nested list as it is created in the ImageSet-tutorial?
It's nice that the capture.uuid is the file name of the newly created tifs but I don't have any orientation anymore in the new pictures folder without hint of the original file name like "IMG_0023.tif"
Excample:
Columns: ['capturename', 'timestamp', 'latitude', 'longitude', 'altitude', 'capture_id', 'dls-yaw', 'dls-pitch', 'dls-roll', 'irr-475', 'irr-560', 'irr-668', 'irr-840', 'irr-717']

The challenge might be that the new tif does not contain the band name. But that would not be so important. The stacked image can also be named like "IMG_0023_0.tif". I tried it with the exiff-data but wasn't successful.

- b) The newly created stack does not contain all six pictures of an Altum capture. How do I add all?
In that case it would be nice to resize the thermal image to the pixel size of the five other band pictures...

- c) my stacked tif ist not aligned. I can see all three colors next to each other. I used the code from the tutorial. Any idea why this happens?

- d) If the whole the skripts of the mentioned toutorial (ImageSet) once works: is the result as follows: undistorted, aligned, unwarped, if available reflectance panel calibrated, radiance corrected and cropped to one GeoTiff? Or do I have still to install the code for alignment, panel calibration etc...?

Healthy regards,
Swawa

poynting, 2020-04-17:
With respect to the capture ID as the filename. It's a little disorienting, I know, but I use this script regularly to process multiple flights worth of data where there are multiple sets of data with the same filenames and UUID ensures there are no name collisions while still putting all the stacks

a) Yeah, you'll need to the capture.image[0].path to the list.  You want the full path though, due to the same reason as above. I had this in one version of the code but I don't think I ever committed it.

b) You need to run alignment first, then copy-paste the warp_matrices to the batch processing code.  If you want to skip this step you can set warp_matrices to None and I believe it will just use the rig relatives, but the alignment won't be as good. https://micasense.github.io/imageprocessing/Batch%20Processing.html#Define-which-warp-method-to-use

c) The Batch processing notebook, when run to completion, results in a set of gps-tagged tiff files which are cropped. I then load them into Agisoft or Pix4D to create geotiffs.

swawa, 2020-04-17:
..With respect to the capture ID as the filename... yes. I agree. The complete idea is creating a file name like IMG_0023_UUID. So I just need to attach the filename to the UUID rename. I will try to modify my output today with your proposal for solution in a)

Thanks a lot so far.

swawa, 2020-04-24:
I added
filename = capture.image[0].path, 
df = pd.DataFrame.from_records(filename, data, index='timestamp', columns=columns)

But the response is 
TypeError: 'module' object is not subscriptable
Did I missunderstood the position to call and add the filename to the list?

swawa, 2020-04-27:
Hi, 
maybe there is an alignment bug..?
in the alignment tutorial are the code lines:

> [...]
> ## Alignment settings
> match_index = 1 # Index of the band 
> max_alignment_iterations = 10
> warp_mode = cv2.HOMOGRAPHY # MOTION_HOMOGRAPHY or MOTION_AFFINE. For # Altum images only use HOMOGRAPHY
> [...]

Last line says, that I have to use only HOMOGRAPHY for Altum
but python tells:
AttributeError: module 'cv2' has no attribute 'HOMOGRAPHY'

So what to use for Altum images?

poynting, 2020-04-27:
cv2.MOTION_HOMOGRAPHY

The example uses this setting to process altum images.

https://micasense.github.io/imageprocessing/Alignment.html

swawa, 2020-04-28:
ok, thanks. This was a missunderstanding. cv2 does not even provide just "HOMOGRAPHY"... I could have known...

The problem of adding the file names to the UUID renaming process persists:
I added to my main script your proposal and the part out of the ImageSets-tutorial:

filename = capture.image[0].path,
df = pd.DataFrame.from_records(filename, data, index='timestamp', columns=columns)

But the response is
TypeError: 'module' object is not subscriptable
Did I missunderstood the position to call and add the filename to the list? Or do I have to add this line to the imageset.py script?

swawa, 2020-04-30:
just found that it must be **capture.images[0].path** instead of capture.image[0].path

poynting, 2020-07-02:
`Image.path` is defined in the `Image` object at https://github.com/micasense/imageprocessing/blob/master/micasense/image.py#L68

`Capture.images` is a list of `Image` objects within a capture.

So if you have a `Capture` object called `cap` you should be able to call

`cap.images[0].path`

to get the fully resolved path of the first image file backing that Image object.

Bartoszko, 2021-07-05:
For me, this solution does not work. I get the message:

AttributeError Traceback (most recent call last)
<ipython-input-4-d31d9fa69478> in <module>
     74 
     75 data, columns = imgset.as_nested_lists()
---> 76 filename = capture.images[0].path
     77 df = pd.DataFrame.from_records(filename, data, index='timestamp', columns=columns)
     78 

AttributeError: module 'micasense.capture' has no attribute 'images'

also the above-mentioned solutions too. Justin do  you have any other suggestions to solve this ORIGINAL-FILE NAMES problem?

Bartoszko, 2021-07-06:
It is important for my, and i suppose others user's, because with original names we have opportunity to use for example PPK/RTK data in external application. Of course after processing whit micasen imageprocessing libary we can use spatial join function but it can do some unnesesary distorsions in processing of remotesensing data and add exta work. It's much easier to keep the original filenames, although using UUID is a great idea.

Bartoszko, 2021-07-06:
I forgot to mention that I am editing a script that is posted on the site 

https://micasense.github.io/imageprocessing/Batch%20Processing.html

MarioCruz22, 2022-09-15:
@Bartoszko Having the same error here… "AttributeError: module 'micasense.capture' has no attribute 'images'" Found some fix?

swawa, 2022-09-15:
Sorry, we don't use the camera anymore as the picture results were not computable for us regarding our needs.

poynting, 2022-09-15:
@Bartoszko filenames should not be necessary to align PPK/RTK data. The images have timestamps, which are in order of time even if the camera didn't have GPS data, and should be *very close* to the PPK timestamps if the camera had GPS data and a PPS pulse (which would always be the case if using DLS2).  If there is ever a need to process multiple flights of data together it is best not to use filenames.  

The only thing I used filenames for was to look up a specific capture to open the file if I needed to look at the image.

In any case I suspect in the code error above the error message is pointing to the problem: when writing ` capture.images[0].path` in your code, "capture" is being interpreted as the module `micasense.capture` not the capture object you are trying to reference.  

Note that I don't work at MicaSense any more since March, and I don't have commit privileges to this repository or access to any data other than what's here already, so it's pretty unlikely I'll be able to help much.

This code adds filename as a property in the list (replace at https://github.com/micasense/imageprocessing/blob/master/micasense/imageset.py#L189).  Works for me, YMMV, best of luck!
```
    def as_nested_lists(self):
        columns = [
            'timestamp',
            'latitude','longitude','altitude',
            'capture_id',
            'dls-yaw','dls-pitch','dls-roll','filename'
        ]
        irr = ["irr-{}".format(wve) for wve in self.captures[0].center_wavelengths()]
        columns += irr
        data = [] 
        for cap in self.captures:
            dat = cap.utc_time()
            loc = list(cap.location())
            uuid = cap.uuid
            dls_pose = list(cap.dls_pose())
            irr = cap.dls_irradiance()
            filename = cap.images[0].path
            row = [dat]+loc+[uuid]+dls_pose+[filename]+irr
            data.append(row)
        return data, columns
```
[image]

MarioCruz22, 2022-09-16:
I was trying to get the alignment script running but it ends up that it doesnt support RedEdge P.. wich is the model i am working with.. Are there some version of RedEdge P i can safely allow to continue the horizontal irradiance válid "test" described in the image?
[image]

MarioCruz22, 2022-09-16:
Getting in trouble with metadata differences like Vignette and Radiance and so on... Anyone knows how to fit RedEdge-P into the tutorial available at https://micasense.github.io/imageprocessing/Alignment.html? Or any other tutorial that can teach me how to treat RedEdge-P data and align its bands?

RedEdge
Vignetting Center               : 639.29433505013424, 480.39791730098648
Vignetting Polynomial           : -0.00035899158967688416, 3.8849091850333786e-06, -2.057088751909051e-08, 5.0152576116649375e-11, -5.827880227440714e-14, 2.5353631877162346e-17

RedEdge-P
Vignetting Polynomial 2D Name   : 0,0,0,1,0,2,0,3,0,4,0,5,1,0,1,1,1,2,1,3,1,4,2,0,2,1,2,2,2,3,3,0,3,1,3,2,4,0,4,1,5,0
Vignetting Polynomial 2D        : 0.467640,0.501446,-0.288309,-0.448964,0.306178,-0.083137,1.384667,0.284610,-0.472609,0.215251,-0.015028,-0.951077,-0.142211,0.523393,-0.179298,-0.884722,-0.275812,-0.063004,0.511571,0.124509,-0.058204