micasense/imageprocessing #115: Issues with set up

aewindle110, 2020-07-28:
I am trying to use these notebooks on Windows OS. I've followed the set up instructions and have successfully imported all required libraries and executed exiftool, but this snippet of code is still not working: 
```
from micasense.image import Image
imagePath = os.path.join('.','data','0000SET','000')

imageName = glob.glob(os.path.join(imagePath,'IMG_0000_1.tif'))[0]

img = Image(imageName)
img.plot_raw(figsize=(8.73,8.73));

from micasense.panel import Panel
panel = Panel(img)
if not panel.panel_detected():
    raise IOError("Panel Not Detected! Check your installation of pyzbar")
else:
    panel.plot(figsize=(8,8));

print('Success! Now you are ready for Part 1 of the tutorial.')
```
The error is:
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-10-0cf0f49199bc> in <module>
      4 imageName = glob.glob(os.path.join(imagePath,'IMG_0000_1.tif'))[0]
      5 
----> 6 img = Image(imageName)
      7 img.plot_raw(figsize=(8.73,8.73));
      8 

S:\Micasense_WQ\imageprocessing-master\imageprocessing-master\micasense\image.py in __init__(self, image_path, exiftool_obj)
     70 
     71         if self.meta.band_name() is None:
---> 72             raise ValueError("Provided file path does not have a band name: {}".format(image_path))
     73         if self.meta.band_name().upper() != 'LWIR' and not self.meta.supports_radiometric_calibration():
     74             raise ValueError('Library requires images taken with RedEdge-(3/M/MX) camera firmware v2.1.0 or later. ' +

ValueError: Provided file path does not have a band name: data\0000SET\000\IMG_0000_1.tif
```

My pwd is: 'S:\\Micasense_WQ\\imageprocessing-master\\imageprocessing-master'
Is the \\ messing things up?

poynting, 2020-07-28:
Usually this is a git-lfs install issue, please see: https://github.com/micasense/imageprocessing/issues/107#issuecomment-596641874

aewindle110, 2020-07-29:
Hi, that was it! I installed git-lfs and re-cloned the repo and that snippet of code ran. Thanks so much.

closed 2020-08-20 by aewindle110