micasense/imageprocessing #117: Modified parameters for "Alignment.ipynb"

kenta0514jp, 2020-08-10:
I would like to hear your experiences or advices to find good parameters for "Alignment.ipynb".  

My experience using RedEdge3 (with images from a building window) was following.  

(1) It was not successful with original parameters.  
max_alignment_iterations = 10
warp_mode = cv2.MOTION_HOMOGRAPHY 
pyramid_levels = 0 # for images with RigRelatives, setting this to 0 or 1 may improve alignment

[image]

(2) I tried many sets of parameters.  Now I found that following parameters works OK for my RedEdge3.  
max_alignment_iterations = 50, pyramid_levels = 2 

[image]
[image]

I also find that the parameters of 
max_alignment_iterations = 10, pyramid_levels = 3 

works OK too. 

Could you tell your experience or advice to find good parameters?

poynting, 2020-10-15:
It's possible that your RedEdge3 does not have RigRelatives information. MicaSense support (support@micasense.com) can help to update your camera's calibration so that it adds this information. Without the rig relatives calibration, a higher pyramid level is necessary. 2 or 3 usually works OK.

kenta0514jp, 2020-10-25:
Thank you very much for your response.  You are right.  My RedEdge3 does not have RigRelatives at that time.  I have updated the firmware and now it is with RigRelatives.  

It is good to know that pyramid level of 2 or 3 usually works OK for images without the RigRelatives.  That information would be useful to process my old images.

closed 2020-10-25 by kenta0514jp