micasense/imageprocessing #120: [Feature Request] Metadata diagnostics after hard landing

and-viceversa, 2020-09-04:
Recently one of our Altums suffered from a hard landing. The camera is mounted so that it does not take a direct impact, however there is still reasonable concern that the transfer of force effects its internal calibration. Image processing (and diagnostic graphs) from this library and Agisoft Metashape all appear normal. Unwarping, alignment, reflectance results, and automatic panel detection all appear to be working normally.

My question is what other diagnostics, image metadata, processing outputs, etc can be tested to help determine whether or not the Altum is functioning as intended? The goal is to avoid a situation where everything "looks good", but under the hood there are problems and unintended consequences.

I understand MicaSense's policy on hard landings, but more guidance in this area would be helpful to diagnose issues at the user level to see if we need to purchase another camera. I am happy to submit PRs based on this discussion.

poynting, 2020-09-04:
Hi, sorry you had a hard lansing thats always a huge concern. Usually the cameras are fine if they look good after visual inspection and process ok, but it's always good to reach out to support@micasense to see what your options are. Unfortunately we can't use this forum as a support forum though so it's best to reach out there.

I doubt there is anything we can or should add in the library to handle this specific case as any diagnostics that might be run on the camera require specialized lab equipment.

and-viceversa, 2020-09-05:
Understood that lab equipment may be required here. My hope was that some aspect of future flight metadata might have non-obvious indications that the camera needs more attention. It looks like MicaSense support is the way to go here.

Thank you as always for the quick reply.

closed 2020-09-05 by and-viceversa