micasense/imageprocessing #122: One image from imageset

janezlapajne, 2020-10-15:
Hello, I went through all available tutorials and everything works fine. One thing is left unanswered for me though. I am wondering does the library includes tool that can generate one big picture from whole imageset? In other words - is somewhere in the code implemented alignment between separate successive captures? I just wanted to ask this to ease my further work :)

poynting, 2020-10-15:
Hello. This library does not provide any image stitching capability. For that you will need to use a commercial program such as Pix4DMapper or Agisoft Metashape, or a open tool like WebODM.

> ### Batch Processing Example
> In this example, we use the micasense.imageset class to load a set of directories of images into a list of micasense.capture objects, and we iterate over that list saving out each image as an aligned stack of images as separate bands in a single tiff file each. Next, we use the metadata from the original captures to write out a log file of the captures and their locations. Finally, we use exiftool from the command line to inject that metadata into the processed images, **allowing us to stitch those images using commercial software such as Pix4D or Agisoft.**

closed 2020-10-15 by poynting

janezlapajne, 2020-10-26:
Thank you for your answer, but I would like to automate a proces of aligning with python script.
I am tryng to align images with gdal tool, but did not succed. It seems that gdal dont know how to read matedata from images?
Do you may have a suggestion for best way how to do it in python?

poynting, 2020-10-27:
I believe [WebODM ](https://github.com/OpenDroneMap/WebODM) (Open Drone Map) has a python interface, and they have recently added support for multispectral data. GDAL doesn't have any image stitching tools (only the ability to overlay images that have already been aligned to each other).

janezlapajne, 2020-10-27:
Hello, thank you for your answer, I will look into it. The reason I started looking into GDAL is that a lot of people recommend it even for mosaicing. I found about merge.py script [doc](https://gdal.org/programs/gdal_merge.html). I saw that it can be done even if images overlap. Do you know anythink about this?

poynting, 2020-10-27:
Yes, I have used if before. It does not do what you are looking for.
However the other tools I have recommended do.

janezlapajne, 2020-10-27:
Aha ok. Thank you very much!

janezlapajne, 2020-10-30:
Hey another question or favor actually. I succesfully enabled webODM with python and successfully enabled mosaicing on raw images. The problem now occures when I do the calibration and extract metadata from raw images and insert that metadata in calibrated images. A lot of values in metadata is missing, beacuse with exiftool (as demonstrated in example: batch_processing) I only write just few ones. I am wondering do you have extended version where whole metadata from raw images would be imported into calibrated images. That way I am sure that I will properly do mosaicing with webODM. Thank you!