micasense/imageprocessing #123: Firmware bellow 2.1.0

janezlapajne, 2020-10-19:
Hello, I need to process some old images. Firmware of a camera at that time was: MicaSense RedEdge firmware version: v1.5.8. I know that in description of this library is stated, that the firmware must be above 2.1.0, but I just wanted to make sure -  is there a way to use this library anyways? :)

barkermi, 2021-02-01:
@janezlapajne any luck with this? I am in a similar situation (RedEdge firmware v1.4.5) and have not been able to adapt the alignment script to work with a legacy set of close range imagery

janezlapajne, 2021-02-01:
Hey, no I stopped trying after a few attempts. I guess there is a reason why the code provided raises an error when the firmware is below the version specified?  What exactly are you trying to do? Do you just want to align images or also convert them to reflectances?

barkermi, 2021-02-01:
I am trying to align captures of seedlings, ideally converting to reflectance as these captures are part of a multitemporal dataset. The align_capture function requires a capture object, and I cannot seem to run the code snippet to produce the object due to old firmware. Running the alignment script on a newer dataset of my Altum imagery throws no errors.

janezlapajne, 2021-02-01:
I think that modification of code for alignment requires minimal efford. I am not so sure about conversion to reflectance. Maybe someone from micasense can help you with that :)

poynting, 2021-04-19:
Your best option is to follow the approach that is taken [here](https://support.micasense.com/hc/en-us/articles/360006368574-Modifying-older-collections-for-Pix4Dfields-support) but with the other tags that are missing.  It will be necessary to update your camera firmware, take an image, and then check which tags are present in the new firmware and copy those to the old images.  It will be necessary to add more tags than the ones described in the support document.

This library requires the radiometric calibration tags present after v2.1.0 and does not work with images that don't contain the appropriate metadata.