micasense/imageprocessing #129: Why is reflectance calculated with math.pi/irradiance?

patrickcgray, 2020-11-13:
One [line 265 of image.py](https://github.com/micasense/imageprocessing/blob/master/micasense/image.py#L265) reflectance is calculated from radiance via `self.__reflectance_image = self.radiance() * math.pi / irradiance`. I'm curious why math.pi is used instead of just dividing radiance by irradiance from the DLS/panel? Is this because the irradiance is a full hemisphere? I don't quite understand the physical/geometric basis of this conversion.

patrickcgray, 2020-11-13:
Okay thanks! But what about when the irradiance values come from the DLS? Would it need to also be pi/irradiance? Or is the `horizontal_irradiance` that comes from the DLS somehow analogous to the panel measurement and needs the same treatment?

fdarvas, 2020-11-13:
The DLS is calibrated to measure irradiance - unlike the camera pixels, which measure light coming from a narrow cone, the DLS integrates light from all directions, i.e. the whole hemisphere. If we assume a cosine response and constant radiance (for simplicity sake) and integrate the radiance over the hemisphere for the DLS we have DLS_output = Radiance * Integral_0_2*pi dphi Integral_0_pi/2 cos(psi)*sin(psi) dpsi = pi*Radiance. So the DLS already measures Irradiance (the factor pi is "built" into these measurements), while the camera measures radiance .

patrickcgray, 2020-11-16:
Okay and because pi is baked into the measurement, in order to cancel it out we need to use the `rad*pi / irradiance` rather than just `rad/irradiance` to correctly calculate reflectance from the DLS measurements?

fdarvas, 2020-11-16:
Correct.

closed 2021-03-12 by patrickcgray