micasense/imageprocessing #130: Problem running the testing installation snippet

JoseVillanuevaMolina, 2020-11-18:
Hello I got an error after running the testing installation snippet.  Any idea?
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
~\anaconda3\envs\micasense\lib\site-packages\pyzbar\zbar_library.py in load()
     57         try:
---> 58             dependencies, libzbar = load_objects(Path(''))
     59         except OSError:

~\anaconda3\envs\micasense\lib\site-packages\pyzbar\zbar_library.py in load_objects(directory)
     53             ]
---> 54             libzbar = cdll.LoadLibrary(str(directory.joinpath(fname)))
     55             return deps, libzbar

~\anaconda3\envs\micasense\lib\ctypes\__init__.py in LoadLibrary(self, name)
    441     def LoadLibrary(self, name):
--> 442         return self._dlltype(name)
    443 

~\anaconda3\envs\micasense\lib\ctypes\__init__.py in __init__(self, name, mode, handle, use_errno, use_last_error)
    363         if handle is None:
--> 364             self._handle = _dlopen(self._name, mode)
    365         else:

OSError: [WinError 126] No se puede encontrar el módulo especificado

During handling of the above exception, another exception occurred:

OSError                                   Traceback (most recent call last)
<ipython-input-2-d0c28ee54db3> in <module>
      3 import os, glob
      4 import numpy as np
----> 5 import pyzbar.pyzbar as pyzbar
      6 import matplotlib.pyplot as plt
      7 import mapboxgl

~\anaconda3\envs\micasense\lib\site-packages\pyzbar\pyzbar.py in <module>
      5 from .locations import bounding_box, convex_hull, Point, Rect
      6 from .pyzbar_error import PyZbarError
----> 7 from .wrapper import (
      8     zbar_image_scanner_set_config,
      9     zbar_image_scanner_create, zbar_image_scanner_destroy,

~\anaconda3\envs\micasense\lib\site-packages\pyzbar\wrapper.py in <module>
    141     c_int,
    142     c_uint_p,    # major,
--> 143     c_uint_p,    # minor
    144 )
    145 

~\anaconda3\envs\micasense\lib\site-packages\pyzbar\wrapper.py in zbar_function(fname, restype, *args)
    134     """
    135     prototype = CFUNCTYPE(restype, *args)
--> 136     return prototype((fname, load_libzbar()))
    137 
    138 

~\anaconda3\envs\micasense\lib\site-packages\pyzbar\wrapper.py in load_libzbar()
    113     global EXTERNAL_DEPENDENCIES
    114     if not LIBZBAR:
--> 115         libzbar, dependencies = zbar_library.load()
    116         LIBZBAR = libzbar
    117         EXTERNAL_DEPENDENCIES = [LIBZBAR] + dependencies

~\anaconda3\envs\micasense\lib\site-packages\pyzbar\zbar_library.py in load()
     58             dependencies, libzbar = load_objects(Path(''))
     59         except OSError:
---> 60             dependencies, libzbar = load_objects(Path(__file__).parent)
     61     else:
     62         # Assume a shared library on the path

~\anaconda3\envs\micasense\lib\site-packages\pyzbar\zbar_library.py in load_objects(directory)
     52                 for dep in dependencies
     53             ]
---> 54             libzbar = cdll.LoadLibrary(str(directory.joinpath(fname)))
     55             return deps, libzbar
     56 

~\anaconda3\envs\micasense\lib\ctypes\__init__.py in LoadLibrary(self, name)
    440 
    441     def LoadLibrary(self, name):
--> 442         return self._dlltype(name)
    443 
    444 cdll = LibraryLoader(CDLL)

~\anaconda3\envs\micasense\lib\ctypes\__init__.py in __init__(self, name, mode, handle, use_errno, use_last_error)
    362 
    363         if handle is None:
--> 364             self._handle = _dlopen(self._name, mode)
    365         else:
    366             self._handle = handle

OSError: [WinError 126] No se puede encontrar el módulo especificado

JoseVillanuevaMolina, 2020-11-18:
If I comment the code #import pyzbar.pyzbar as pyzbar, the error disappears. 
[image]

poynting, 2020-11-18:
Seems like the libzbar library isn't properly installed.

What OS?
What actions have you taken?
Were there any errors during those actions?

JoseVillanuevaMolina, 2020-11-23:
My OS is Windows 10. I downloaded the environment and ran the setup test code.

philipperemy, 2021-11-14:
One dep is missing in the conda requirements.

https://anaconda.org/conda-forge/zbar

philipperemy, 2021-11-14:
@poynting can you add it to the YML file? 
It worked after installing it.

philipperemy, 2021-11-14:
You can either install it at the OS level or in conda.

poynting, 2021-11-14:
It's been a while, but last time I tried it didn't work across all
platforms from conda.

If someone wants to test it on Win/Linux/Mac and make a PR, I'll merge it.

philipperemy, 2021-11-14:
@poynting I guess you can make it work for everyone in the easiest way, I guess you can add it as a dep to conda (rather than relying on a manual installation such as brew).

poynting, 2021-11-14:
Yes, I understand the idea. The last time I tested doing it that way, it
did not work. I don't recall the specific issue, but I believe it was on
Mac or Linux where it failed.

If someone will confirm that it does work now across platforms, and update
the notebooks and documentation, I will be happy to merge the PR. Otherwise
it will need to wait until I have some time in the office to validate
across platforms.