micasense/imageprocessing #131: Micasense as a ground sensor

khunsafatima, 2020-11-25:
Hi, I have used Micasense RedEdge to capture trees at the ground from a distance of 10m. As expected, its five bands have some spatial shift from each other. When I use the code provided by missense to align the bands, I get the following error:

_ValueError: Input must be >= 2-d._

Kindly suggest how can I fix this issue.

poynting, 2021-04-19:
I haven't seen this issue before. Can you share sample data (a full capture) that shows this issue?

poynting, 2021-05-24:
Closing due to lack of response.

closed 2021-05-24 by poynting

mirob9363, 2021-10-29:
I think it could be due to something with the calibration panel, for I get this error when running the panel through the "MicaSense Image Processing Setup" Tutorial.
[image]

poynting, 2021-11-07:
> I have the same issue.
> 
> In panel.py, the function _radiance_ calls the _region_stats_, which calls the _np.fliplr_, which raises the error:
> 
> _ValueError: Input must be >= 2-d._
> 
> The issue is that the _self.serial_ is None because it isn't initialized with a value since the _self.image.panel_serial_ is None as well.
> 
> Here can be found the images I am using and the code (execute align_multispectrals_fast.py):
> 
> https://drive.google.com/drive/folders/1IPgvsvwEajhYcFJVFshbRjrnfEBDl4US?usp=sharing
> 
> Please help me,
> 
> Thank you in advance.

@mpafit02 
The Alignment notebook seems to run fine with this data. Since there is no panel image, the panelNames must be set to None.

[image]

poynting, 2021-11-07:
@mirob9363 
It's difficult to help without the problem data. However, in the picture above, it appears there is a specular reflection right on the QR code on the panel. This might be the problem, as this would make the QR code difficult for the software to find.  I would suggest trying a different panel image which doesn't contain the reflection of the sun on the QR code.  I like to use the Panels notebook to debug panel issues, but as the exception is showing, the panel is not detected in this image.

[image]