micasense/imageprocessing #134: Error Install micasense imageprocessing

janse28, 2020-12-17:
Hello, 

I would like to work to with your skript and install Anaconda 3. In fewer issues I read that a downgrading python version is necessary to run this skript. Therefore I installed on Anaconda python version 3.6.12.

Nevertheless I get an error

(base) C:\Users\XX>python --version
Python 3.6.12 :: Anaconda, Inc.
(base) C:\>cd imageprocessing
(base) C:\imageprocessing>activate micasense
(micasense) C:\imageprocessing>pytest
ImportError while loading conftest 'C:\imageprocessing\tests\conftest.py'.
tests\conftest.py:29: in <module>
    import micasense.capture as capture
micasense\capture.py:30: in <module>
    import micasense.image as image
micasense\image.py:36: in <module>
    import micasense.metadata as metadata
micasense\metadata.py:28: in <module>
    import exiftool
E  ModuleNotFoundError: No module named 'exiftool'

The exiftool (version 12.12) I puted on C:\ and made a path in my Environment Variables for my Account.
Thanks for your efforts,

Jan

poynting, 2021-04-19:
Jan, do you still have this issue or did you come up with a resolution? I just built a new environment from the main branch and it worked fine (though it takes much longer than before to resolve packages).

poynting, 2021-07-13:
Closed due to inactivity.

closed 2021-07-13 by poynting