micasense/imageprocessing #135: CondaEnvException: Pip failed

FalconMath1975, 2020-12-30:
At the end of executing conda env create -f micasense_conda_env.yml : 
Pip subprocess error:
  WARNING: Generating metadata for package pyexiftoolpy produced metadata for project name pyexiftool. Fix your #egg=pyexiftoolpy fragments.
ERROR: Requested pyexiftool from git+https://github.com/smarnach/pyexiftool.git#egg=pyexiftoolpy (from -r d:\Ag Imagery\condaenv.rrxe8wh6.requirements.txt (line 3)) has different name in metadata: 'PyExifTool'

failed

CondaEnvException: Pip failed

FalconMath1975, 2020-12-31:
used code in HRRMXJ 's pull request to successfully solve issue.

songliVT, 2021-03-01:
How do you clone from this pull request?

poynting, 2021-04-19:
Fixed via @HRRMXJ's PR.

closed 2021-04-19 by poynting