micasense/imageprocessing #136: Results of Batch Processing Tutorial can't be processed by Pix4DFields

Tobias-UniBwM, 2021-01-07:
Hello, first of all thank you very much for your efforts you put into this library. It is very helpful to understand the general workflow of processing multispectral imagery.

I've got an Altum sensor and the Pix4Dfields software for image processing. Sadly Pix4Dfields doesn't do any image alignment so it's not difficult to spot artifacts in the generated orthomosaic. An obvious solution to this is to align all captures beforehand and then feed them to Pix4Dfields. So I followed those steps you described in the [Batch Processing Tutorial](https://micasense.github.io/imageprocessing/Batch%20Processing.html). Unfortunately the stacked tiff files aren't getting processed by Pix4Dfields which says "Your camera model isn't supported". I checked the new files with exiftool and all tags mentioned in the tutorial had been properly transferred.

Is there a solution to this? Also I keep wondering, how would Pix4Dfields or any other software know which one of the stacked images does belong to a specific spectral band? Which is necessary to calculate any index image, as you know of course. Additionally why can't I just align all the raw captures in advance (before radiometric calculations, etc.) and just copy all the metadata from every unaligned image to the aligned image? Following this way Pix4Dfields had no chance to detect any difference between aligned and unaligned images except for the inconsistent resolution across all captures due to the alignment procedure.

poynting, 2021-01-07:
It may be best to contact pix4d support. Pix4d fields does image alignment,
they may be able to help work through the appropriate settings. There is a
rig alignment mode that takes a little longer but will find a better
alignment if you are seeing bad results.

Tobias-UniBwM, 2021-01-08:
Thanks for your fast reply. I found out that rig alignment is only supported for non thermal images in Pix4Dfiels. I tried without thermals and the results seem to be much better now, thank you for your hint!

One other thing if I may. It's a mess to manage the images' exif data in python even with exiftool since a lot of tags are unknown to it. Is there a simpler way to transfer metadata from one image to another without studying exif and tif specifications? How do you initially write the metadata to the images?

poynting, 2021-01-08:
Without writing low level code, the exiftool or exiv2 applications are your
best bets.

Tobias-UniBwM, 2021-01-11:
Ok, well, it is what it is. Nevertheless, thank you very much for your time and help!

closed 2021-01-11 by Tobias-UniBwM