micasense/imageprocessing #137: QUESTION: 64-bit image output

hjken, 2021-01-25:
I modified Tutorial 1 to get reflectance images for an image set but when I tried to input them into another program (WebODM) to create an orthomosaic I got an error that the program does not accept 64-bit images. I was surprised because the original images (from MS RedEdge) are 16 bit. So, to check, I saved the converted reflectance image (tiff) created in Tutorial 1. It also is a 64-bit image. Could someone explain why the output of Tutorial 1 is a 64-bit image when Tutorial 1 is just converting DN values to reflectance values, it is not creating a stacked image?

poynting, 2021-04-18:
The `flightReflectanceImage` array is double precision floating point. The output format of the image will depend on how it is written to a file.

If using GDAL to save out the reflectance image, it is possible to define the output type of the resulting TIFF.  For example in the [save capture function](https://github.com/micasense/imageprocessing/blob/8190748bd200d4e37c74b44202690421d7d64f85/micasense/capture.py#L379), we write a 16-bit TIFF, where 32768 represents 100% reflectance (to provide some overhead for saturation in the case of specular reflections).

poynting, 2021-07-13:
Closed due to inactivity.

closed 2021-07-13 by poynting