micasense/imageprocessing #140: LWIR alignment parameters

swawa, 2021-02-09:
Our camera operates with the latest firmware, but setting pyramid_levels to zero results in a real bad alignment. Setting pyramid_levels to 3 is better but not good enough for pixelwise calculations e.g. as needed for NDVI calculations. 
Therefore I calibrated our Altum geometrically using OpenCV and the chessboard method. With the gained values I get better results in the alignment. For the LWIR-band the chessboard method is not possible. Can you tell me which values are used in the default alignment for the LWIR-band?
Thanks

poynting, 2021-04-18:
Can you upload some complete captures you're having trouble with? If not here, then email support @ micasense and they can help you out.

poynting, 2021-07-13:
Closed due to lack of response.

closed 2021-07-13 by poynting