micasense/imageprocessing #146: OpenCV Install conflict between Conda .yml and setup.py

and-viceversa, 2021-05-02:
`micasense` library fails with a QT [plugin error](https://stackoverflow.com/questions/41994485/how-to-fix-could-not-find-or-load-the-qt-platform-plugin-windows-while-using-m) if you install `micasense` into a conda environment that requires `opencv`, and attempt to call `.imshow()` or other plotting functions.

`micasense` conda env has `opencv` as a dependency, while the `setup.py` installer uses `opencv-python`.

Cause: `opencv-python` [install instructions](https://pypi.org/project/opencv-python/) states:

> There are four different packages (see options 1, 2, 3 and 4 below) and you should SELECT ONLY ONE OF THEM. Do not install multiple different packages in the same environment. There is no plugin architecture: all the packages use the same namespace (cv2). If you installed multiple different packages in the same environment, uninstall them all with pip uninstall and reinstall only one package.

Since this is a namespace issue, it also implies that installing `opencv` alongside `opencv-python` will cause issues. This is also complicated by the fact that `opencv` is a conda install, and `opencv-python` is a `pip` install and the two do not respect each other's actions. This is pernicious, since the QT plugin error messages don't hit that this is the issue.

Is there a reason for using both `opencv` and `opencv-python` in their respective places?

poynting, 2021-05-03:
I think this is mainly a fallout of the history that this code started as a loose group of scripts intended to be a tutorial, but has morphed over time with more usage as a library and a lot more higher level functionality (like imageset/batch processing).

There may need to be a higher level overhaul that splits what's here into two separate repositories, or at least reorganizes things a little more in line with the current use cases.

One option would be to split the notebooks and tutorials into another repository. Alternatively, we could move the notebooks into a `tutorials` or `notebooks` folder, as they are still a good way to exercise and demonstrate library code.

However, one major pain point is that MicaSense support gets a lot of questions (even thought it's stated they don't support this code) and does work to help as much as possible, so the easier to install the better. The reason we went with `conda` in the first place was that it was the only reliable way we could get the code to work on mac/linux/windows, which is a core requirement for our team.  Having a conda package that installed the core library and pulled dependencies would be a welcome change.

and-viceversa, 2021-05-03:
This issue is somewhat obscure, but definitely a difficult case for most users. It looks like it can be pinned as part of a general refactor.

I'm grateful for this library - happy to incrementally commit documentation and additions I've found useful. e.g. #148

Thank you as always for the thorough response.