micasense/imageprocessing #150: Use Exiftool from the command line to write metadata to images

btpearse, 2021-05-23:
I'm stuck on the last step of the Batch Processing notebook at the "Use Exiftool from the command line to write metadata to images". When I run the code it doesn't seem to change anything in my tif files.

I found a similar issue here: https://github.com/micasense/imageprocessing/issues/121, but it looks like it never got solved. 

I run the following code and get an output of 0
```
import subprocess

if os.environ.get('exiftoolpath') is not None:
    exiftool_cmd = os.path.normpath(os.environ.get('exiftoolpath'))
else:
    exiftool_cmd = 'exiftool'
        
cmd = '{} -csv="{}" -overwrite_original {}'.format(exiftool_cmd, fullCsvPath, outputPath)
print(cmd)
subprocess.check_call(cmd)
```
Very new to this stuff so appreciate the help.

poynting, 2021-05-24:
I've noticed that sometimes it does that, and I'm not sure what the issue is. Try to copy/paste the command out into a new anaconda command line and see if that works.

btpearse, 2021-05-24:
I may need some guidance for how to do that. This is what I tried:

- open the Anaconda prompt
- set the cd and environment to what I was running in jupyter notebook
- start python
- paste the command line in

I imagine there's a better way of doing it because this didn't work for me!

poynting, 2021-05-24:
Sure, it's not a line of python, it's just a command (like the `jupyter notebook .` command).  So you can just copy it into the anaconda prompt - nothing else should be required.  If that doesn't work, post your command and the output here; maybe there's something going wrong with the command itself.

Here's the example for the Batch Processing notebook running:
```
c:\exiftool\exiftool.exe -csv="C:\Users\justi\Downloads\RedEdgeImageSet\0000SET\..\stacks\log.csv" -overwrite_original C:\Users\justi\Downloads\RedEdgeImageSet\0000SET\..\stacks
```
The output is just the number of files found/updated.

btpearse, 2021-05-24:
Interesting, I run it and only one of my tifs is updated. I run it again and another tif gets updated. 
This is my command: 
```C:\exiftool\exiftool.exe -csv="E:\0006SET\stacks\log.csv" -overwrite_original E:\0006SET\stacks```

The output is this long list:
[image]

poynting, 2021-05-24:
That seems to be telling us that there's a disconnect between the files in the directory, and the paths in the log.csv file.  If the filenames between the two don't match those messages will show up.

btpearse, 2021-05-25:
Ok well at least I have the command running now. I will have to do some troubleshooting to see why my file names aren't matching up. Thanks for your help on this!

ghentr, 2022-02-18:
Hi; this is related.  I'm also running the batch processing tutorial, and I get a "No such file or directory" error when I try to run the exiftool -csv line from a Jupyter notebook (running on a Mac).  After trying for 4 hours to forensically investigate this problem (because I'm new to exif), I have determined that it's a problem with Jupyter and not exiftool: I know this because I just successfully updated the metadata for all 355 tiffs in the "stacks" folder using the terminal.

Does anyone have any idea what's causing this strange behavior in Jupyter?

Thanks!

++++++
---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
/var/folders/qt/3lb1lh3x7yz_v0g5s0mnj8xr0000gp/T/ipykernel_69228/2051487202.py in <module>
     14 cmd = '{} -csv="{}" "{}"'.format(exiftool_cmd, fullCsvPath, outPath)
     15 print(cmd)
---> 16 subprocess.check_call(cmd)

~/mambaforge/envs/micasense/lib/python3.7/subprocess.py in check_call(*popenargs, **kwargs)
    356     check_call(["ls", "-l"])
    357     """
--> 358     retcode = call(*popenargs, **kwargs)
    359     if retcode:
    360         cmd = kwargs.get("args")

~/mambaforge/envs/micasense/lib/python3.7/subprocess.py in call(timeout, *popenargs, **kwargs)
    337     retcode = call(["ls", "-l"])
    338     """
--> 339     with Popen(*popenargs, **kwargs) as p:
    340         try:
    341             return p.wait(timeout=timeout)

~/mambaforge/envs/micasense/lib/python3.7/subprocess.py in __init__(self, args, bufsize, executable, stdin, stdout, stderr, preexec_fn, close_fds, shell, cwd, env, universal_newlines, startupinfo, creationflags, restore_signals, start_new_session, pass_fds, encoding, errors, text)
    798                                 c2pread, c2pwrite,
    799                                 errread, errwrite,
--> 800                                 restore_signals, start_new_session)
    801         except:
    802             # Cleanup if the child failed starting.

~/mambaforge/envs/micasense/lib/python3.7/subprocess.py in _execute_child(self, args, executable, preexec_fn, close_fds, pass_fds, cwd, env, startupinfo, creationflags, shell, p2cread, p2cwrite, c2pread, c2pwrite, errread, errwrite, restore_signals, start_new_session)
   1549                         if errno_num == errno.ENOENT:
   1550                             err_msg += ': ' + repr(err_filename)
-> 1551                     raise child_exception_type(errno_num, err_msg, err_filename)
   1552                 raise child_exception_type(err_msg)
   1553 

FileNotFoundError: [Errno 2] No such file or directory: 'exiftool -csv="~/RedEdgeImageSet/stacks/log.csv" "~/imageprocessing/RedEdgeImageSet/stacks/"': 'exiftool -csv="~/imageprocessing/RedEdgeImageSet/stacks/log.csv" "~/imageprocessing/RedEdgeImageSet/stacks/" '