micasense/imageprocessing #151: align images with low latitude(6.5-7ft) for grass

yoonjoo8, 2021-05-24:
Hello. I wonder if there are anyways to align images taken by the distance of 6.5-7ft.
Is it affected by the low latitude and the number of leaves/ types of plants (like weed or rice) or panel_refelctance?
(I have used our panel index provided by micasense (0.5,0.5,0.5,0.5,0.5). 

I have tried to change the value of match_index, iterations, pyramid_levels (0-3), but unfortunately, haven't aligned images very well. 
Images are taken by micasense-mx with 5 bands.

and-viceversa, 2021-05-24:
I haven't tried with such a close distance, but I'd try alignment with Rig Relatives in this case.

Like the tutorials say, image alignment works best with strong man-made features. In their absence (desert, wetlands, etc) Rig Relatives has worked for me.

poynting, 2021-05-24:
Due to the parallax effect, there will always be geometric offsets between imagery. These offsets become more noticeable as the camera gets closer to the target, and are made worse when there are varying distances within the same image; for example, it is difficult to align tall trees and the ground when the camera is close to the tops of the trees, or tall plants against the ground when the camera is close to the tops of the plants.  

For a flat surface from a constant distance, it is possible to get a good alignment, but the alignment will change as the distance changes.  In this case the camera needs to be pointing straight down, and the camera height above the target needs to be controlled very well.  Use identifiable man-made objects (such as some reflectance panels or ground control targets) to get a good alignment, then re-use that alignment for the images with fewer distinct features.

amyyjch, 2021-05-25:
Thank you for your explanation and comment. It is much appreciated. 
I have used like a pole with the camera to maintain the distance and to be pointing straight down. 
And images of the panel were aligned well (I only used one sample though).

However, for the grass with many leaves, it may be due to the parallax effect in the closer distance. 
I will try to capture images with further distance next time. 
Is there any recommendation for the distance to capture images?

poynting, 2021-07-13:
The distance should be around 10x or more the maximum height/depth in the image. So for a plant which is 1m tall, I would not expect to align using these methods closer than 10m or so.  This is just a rule of thumb.  Your results may vary.

closed 2021-07-13 by poynting

wanghanruibaba, 2022-09-13:
Hello brother, is your image aligned now? Close-up images, due to parallax, cause the leaves to be deformed, making it difficult to align. Have you found a way to align it now? Can you share ideas? grateful