micasense/imageprocessing #165: Alignment for a whole directory

sgtkellox, 2021-08-12:
I used your code to make some alignments and I like the result, yet I found it a bit inconvenient that you have to process the images one by one. I made some changes to process directorys of images, it works fine. Shall I submitt the code to you? 
Regards, Felix

poynting, 2021-08-12:
Have you seen the Batch Processing notebook?

XingWei19, 2021-08-15:
> 
> 
> I used your code to make some alignments and I like the result, yet I found it a bit inconvenient that you have to process the images one by one. I made some changes to process directorys of images, it works fine. Shall I submitt the code to you?
> Regards, Felix

Hello Felix, I am working on batch processing RedEdge 5-band images. I would like to test your codes if you are willing to share them.  Thanks, Xing

sgtkellox, 2021-08-15:
I will have a look at that notebook. @XingWei19 ill paste my code here so you can use it. Ill assume that is ok.
Here you go, hope it helps:
'''
import os, glob
import micasense.capture as capture
import cv2
import numpy as np
import matplotlib.pyplot as plt
import micasense.imageutils as imageutils
import micasense.plotutils as plotutils
import imageio

path = path goes here 

os.mkdir(os.path.join(path,"cir_aligned"))
os.mkdir(os.path.join(path,"rgb_aligned"))

files = os.listdir(path)
names = []
for fileName in files:
    if fileName.endswith(".tif"):
        name = fileName.split("_",2)
        if not name[0]+"_"+name[1] in names:
            names.append(name[0]+"_"+name[1])

for name in names:
    panelNames = None

# # This is an older RedEdge image without RigRelatives
# imagePath = os.path.join(os.path.abspath('.'),'data','0000SET','000')
# imageNames = glob.glob(os.path.join(imagePath,'IMG_0001_*.tif'))
# panelNames = glob.glob(os.path.join(imagePath,'IMG_0000_*.tif'))

# # Image from the example RedEdge imageSet (see the ImageSet notebook) without RigRelatives.
# imagePath = os.path.expanduser(os.path.join('~','Downloads','RedEdgeImageSet','0000SET'))
# imageNames = glob.glob(os.path.join(imagePath,'000','IMG_0013_*.tif'))
# panelNames = glob.glob(os.path.join(imagePath,'000','IMG_0000_*.tif'))

# This is an altum image with RigRelatives and a thermal band
    imagePath = os.path.join(path)
    imageNames = glob.glob(os.path.join(imagePath,name+'_*.tif'))
    #panelNames = glob.glob(os.path.join(imagePath,'IMG_0000_*.tif'))

# Allow this code to align both radiance and reflectance images; bu excluding
# a definition for panelNames above, radiance images will be used
# For panel images, efforts will be made to automatically extract the panel information
# but if the panel/firmware is before Altum 1.3.5, RedEdge 5.1.7 the panel reflectance
# will need to be set in the panel_reflectance_by_band variable.
# Note: radiance images will not be used to properly create NDVI/NDRE images below.
    if panelNames is not None:
        panelCap = capture.Capture.from_filelist(panelNames)
    else:
        panelCap = None

    cap = capture.Capture.from_filelist(imageNames)

    if panelCap is not None:
        if panelCap.panel_albedo() is not None:
            panel_reflectance_by_band = panelCap.panel_albedo()
        else:
            panel_reflectance_by_band = [0.67, 0.69, 0.68, 0.61, 0.67] #RedEdge band_index order
        panel_irradiance = panelCap.panel_irradiance(panel_reflectance_by_band)    
        img_type = "reflectance"
        cap.plot_undistorted_reflectance(panel_irradiance)
    else:
        if cap.dls_present():
            img_type='reflectance'
            cap.plot_undistorted_reflectance(cap.dls_irradiance())
        else:
            img_type = "radiance"
[... 37 lines cut ...]
# for cir false color imagery, we normalize the NIR,R,G bands within themselves, which provides
# the classical CIR rendering where plants are red and soil takes on a blue tint
    for i in cir_band_indices:
        im_display[:,:,i] =  imageutils.normalize(im_aligned[:,:,i])

    cir = im_display[:,:,cir_band_indices]
    fig, axes = plt.subplots(1, 2, figsize=figsize)

    gaussian_rgb = cv2.GaussianBlur(rgb, (9,9), 10.0)
    gaussian_rgb[gaussian_rgb<0] = 0
    gaussian_rgb[gaussian_rgb>1] = 1
    unsharp_rgb = cv2.addWeighted(rgb, 1.5, gaussian_rgb, -0.5, 0)
    unsharp_rgb[unsharp_rgb<0] = 0
    unsharp_rgb[unsharp_rgb>1] = 1

    # Apply a gamma correction to make the render appear closer to what our eyes would see
    gamma = 1.4
    gamma_corr_rgb = unsharp_rgb**(1.0/gamma)
    fig = plt.figure(figsize=figsize)
    imtype = 'jpg' # or 'jpg'
    
    imageio.imwrite(os.path.join(path,"rgb_aligned",name+'rgb.'+imtype), (255*gamma_corr_rgb).astype('uint8'))
    imageio.imwrite(os.path.join(path,"cir_aligned",name+'cir.'+imtype), (255*cir).astype('uint8'))

'''

poynting, 2021-08-24:
It's not necessary to align each capture each time.  It's much faster to use one image with good features to find an alignment matrix, then use that to align all of the images from the same capture session.

The batch processing notebook does this, using the output of the alignment notebook.  

The 10-band Batch Processing notebook further improves this by using the `imgset.save_stacks` function which is internally multithreaded.  The 10-band approach can also be used with 5-band or 6-band data.

XingWei19, 2021-09-07:
@sgtkellox Thank you for sharing your codes. I appreciate it.

XingWei19, 2021-09-07:
@poynting When I testing the 10-band Batch Processing notebook, I got an error saying, "ImageSet' object has no attribute 'save_stacks". Do I need to update the library/package to use this function? Thanks

karimelgazar, 2022-02-13:
use `imgset.process_imageset` instead

zekrimohamed, 2022-02-28:
@poynting When I tested the 10-band Batch Processing notebook, I got an error saying, "ImageSet' object has no attribute 'save_stacks". Do I need to update the library/package to use this function? Thanks

YY-99999, 2022-05-12:
Hi @poynting 

I am having same issue "ImageSet' object has no attribute 'save_stacks". I have tried **imgset.process_imageset** and it seemed not working as well. Any ideas ? Many Thanks

alegent, 2022-09-01:
I am having same errors on using both imgset.save_stacks() and imgset.process_imageset() . Any progress on this issue?