micasense/imageprocessing #166: Batch processing: ValueError

XingWei19, 2021-08-15:
Hello micasense community,

I am testing the "Batch Processing.ipynb" on RedEdge five-band drone images. I came across two errors:

Error1: when run the first cell of "Load Images into ImageSet", it says "OSError: No files provided. Check your file paths."  It worked after I modified the imagePath and panelNames parameters.

Error2: when run the third cell of "Load Images into ImageSet", it says "ValueError: Number of class have to be an integer greater than 2 and smaller than the number of values to use". 

Can you help to solve the Error2? Thanks

Detailed error message was below, 
"---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-4-1653a2a531cf> in <module>
     19 
     20 import jenkspy
---> 21 breaks = jenkspy.jenks_breaks(df[color_property], nb_class=num_color_classes)
     22 
     23 color_stops = create_color_stops(breaks,colors='YlOrRd')

~\Anaconda3\envs\micasense\lib\site-packages\jenkspy\core.py in jenks_breaks(values, nb_class)
    105 
    106     if nb_class >= nb_values or nb_class < 2:
--> 107         raise ValueError("Number of class have to be an integer "
    108                          "greater than 2 and "
    109                          "smaller than the number of values to use")

ValueError: Number of class have to be an integer greater than 2 and smaller than the number of values to use"

poynting, 2021-08-24:
It should be possible to use the link in the [ImageSets notebook](https://micasense.github.io/imageprocessing/ImageSets.html) to download the [sample data](https://sample.micasense.com/imageprocessing/RedEdgeImageSet.zip) that is used to generate the notebook, which can verify that the code is working properly.

As far as the value error, it's possible that the data is missing the dls property that the code is trying to classify.  Try printing the `columns` variable before the line that error is on, and try setting `color_property` to one of those values (such as `altitude`).

XingWei19, 2021-09-07:
Hi Justin,

Thank you for the suggested comments. Using the sample data, I successfully run the "Batch Processing.ipynb". However, when I use my own data set, it gave me an error at the line, "panel_irradiance = panelCap.panel_irradiance(panel_reflectance_by_band)". The error says, "ValueError: Input must be >= 2-d." Can you explain what is this step for?

I tried using fixed values for the sample data, as "panel_irradiance = [0.8088253338980584,0.8246040254690193,0.7485704221000492,0.542069532328326,0.6110526527269586]". Then I was able to run all the codes.

Another problem I came across was regarding the warp_matrices. If I use the warp_matrices without RigRelatives, images generated in the "thumbnails" folder had colorful shadows. If I use the warp_matrices as None, most images don't have shadows. Can I ask what is the difference between with and without RigRelatives?

Thanks

poynting, 2021-09-07:
With the `ValueError` my guess is that the panel is not found in all of the images. Try to plot the panels to see if this is the issue.  If panels are not used, make sure `panelNames=None`

```
panelCap.plot_panels()
```

RE Warp Matrices: It is necessary to run the 10-band alignment notebook and use the `warp_matrix` from there. Since there is no metadata to align the different cameras, alignment is required to find the full matrix.

XingWei19, 2021-09-07:
Hi Justin, 

Thank you for the quick responses and explanations. I think I am much more comfortable running the batch processing notebook now. So I will close this open issue. 

Thanks again!

closed 2021-09-07 by XingWei19