micasense/imageprocessing #167: Irradiance calculation in actual code

Camilochiang, 2021-08-24:
Good day, 
- In issue  #129 was explained why to calculate reflectance, radiance need to be multiplied by pi before been divided by irradiance. In your [Tutorial 1](https://github.com/micasense/imageprocessing/blob/master/MicaSense%20Image%20Processing%20Tutorial%201.ipynb) in the other hand this is not done as the values are extracted from the panel. 
- Question: both in image.py  (def reflectance) and panel.py (def irradiance_mean) the irradiance is calculated in that way, without the option to specify from where is coming the radiance value (DLS2 or panel). Is it correct to say that in the case of calculating irradiance from the panel this functions can be modified removing the pi multiplication? 

Thanks!

poynting, 2021-08-24:
In the tutorial, for simplicity, irradiance isn't considered at all.  In this case it's simply that there is a scale factor from pixel values (or, more specifically, linear radiance as computed by the calibration function) to linear reflectance. This scale factor is computed by the average of the panel region. Since we use a radiance image and reflectance values in this computation, the scale factor will have the `pi` factor included.

In image.py and panel.py the *irradiance* may be used to calculate the reflectance if provided to the function, or may come from the image metadata.  We normalize around the units of radiance and irradiance, and convert back and forth as needed.  Therefore when we use a panel, we use the radiance image to calculate the irradiance at the panel, and use that irradiance to convert other radiance images to reflectance images.  This is useful for more complex processing because the output of the DLS is also irradiance, so the functions to compute reflectance from radiance images are already in place.  This also makes it easier when the irradiance comes from another source, like a lab spectroradiometer.

So the normal process is to find the irradiance using a panel image, then apply that irradiance to other images using `image.reflectance(irradiance)`

Camilochiang, 2021-08-24:
Thanks for the quickly answer. Do the code assume that cos there is a DLS present you can use its irradiance values? Im asking as my actual problem is that I want to use the DLS information (i.e. pose) but not the irradiance coming from it as some shadows may be covering it

closed 2021-08-24 by Camilochiang