micasense/imageprocessing #169: Low distance of 10-band Alignment

JhihKai, 2021-10-01:
Hello,

I know Micasense doesn't recommend to capture pictures lower than 30m, but for some reason I've to try this.
I tried to align 10-band images captured with 0.6 and 1m height, but the result is not ideal, the problem of shifting exists.
Reference band, max iterations, warp mode, and pyramids level were changed, the best result is improved, but still.

I read the issue #45 then try to add nol += 2 on line 114 of imageutils.py as poynting mentioned. Unfortunately I don't know how to add it.
[image]
 
Here is the images:
https://drive.google.com/drive/folders/172VaY4RgOVATTexIS8M7L8DvK_ufI8E9?usp=sharing

closed 2021-11-07 by JhihKai

poynting, 2021-11-07:
I was not able to align the 0.6m data.

For the 1m level these settings work fine for me in the Alignment-10Band notebook:
```
## Alignment settings
match_index = 4 # Index of the band, here we use green
max_alignment_iterations = 100
warp_mode = cv2.MOTION_HOMOGRAPHY # MOTION_HOMOGRAPHY or MOTION_AFFINE. For Altum images only use HOMOGRAPHY
pyramid_levels = 5 # for this case we force a 5-level pyramid.
```

[image]
[image]

This seems to work very well, however, the alignment mainly aligns the background (floor). Due to parallax, the specimens on the raised platform are not perfectly aligned. For better results, place the specimens flat against the background.

MarcoVuerich, 2022-05-03:
Hello,

I'm having the same issue of  @JhihKai : try to add nol += 2 on line 114 of imageutils.py.

Thank you,
Marco