micasense/imageprocessing #177: Attribute Error - 'ExifTool' object has no attribute 'getMetadata'

anthea-ops, 2022-03-04:
Hello, new to python and having difficulty; this may be basic and I am just missing something. Please indicate if you need more information. This is the error message I am getting when running the second part of the startup (MicaSense Image Processing Setup) code:

Successfully imported all required libraries.

Successfully executed exiftool.
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
/var/folders/dq/xpxx0g3j4r5gxc2n777_j9l40000gp/T/ipykernel_8445/1712436492.py in <module>
     34 imageName = glob.glob(os.path.join(imagePath,'IMG_0000_1.tif'))[0]
     35 
---> 36 img = Image(imageName)
     37 img.plot_raw(figsize=(8.73,8.73));
     38 

~/VIMS/imageprocessing/micasense/image.py in __init__(self, image_path, exiftool_obj)
     67             raise IOError("Provided path is not a file: {}".format(image_path))
     68         self.path = image_path
---> 69         self.meta = metadata.Metadata(self.path, exiftool_obj=exiftool_obj)
     70 
     71         if self.meta.band_name() is None:

~/VIMS/imageprocessing/micasense/metadata.py in __init__(self, filename, exiftoolPath, exiftool_obj)
     47             raise IOError("Input path is not a file")
     48         with exiftool.ExifTool(self.exiftoolPath) as exift:
---> 49             self.exif = exift.get_metadata(filename)
     50 
     51     def get_all(self):

AttributeError: 'ExifTool' object has no attribute 'get_metadata'

Thank you.

volpatoo, 2022-03-04:
Hello,

I have got the same issue. Well, I am glad to know that more users had the same problem because I was blamed for not being able to run a simple code...

Looking forward to seeing how to fix this problem. Thanks!

mcbridejc, 2022-03-05:
Hi folks, 

I happened to observe your issue, and can tell you that the problem you are seeing is due to changes in the API made in recent releases of pyexiftool by the new maintainer (https://github.com/sylikc/pyexiftool/). 

To fix this issue -- at least in the interim until MicaSense can address it -- I suggest you try reverting to an older version. I think 0.4.13 is the version you want to go back to. To revert, using pip, run `pip install pyexiftool==0.4.13`. If you're using conda, I think `conda install pyexiftool==0.4.13` is what you need (although I don't use conda so ymmv).

volpatoo, 2022-03-05:
Hello guys! 

Thanks, God, for existing such good people in this world! I really appreciate that @mcbridejc. It worked well and now also passed in the test using `pytest -n auto`.

anthea-ops, 2022-03-08:
Thank you so much @mcbridejc

fdarvas, 2022-03-30:
I looked into the the current repository for pyexiftool (https://github.com/sylikc/pyexiftool/) and it looks like the code does not support the 'get_metadata(filename)' method for the Exiftool object anymore - not sure if that is intentional or if that is simply work in progress. 
At the moment sticking with an older version of pyexiftool, e.g. pyexiftool==0.4.13  seems to be the best course of action - I will update setup.py to to require pyexiftool<=0.4.13 for now.

wwolff7, 2022-04-22:
@mcbridejc works for me too. Thanks a lot!

FlLobo, 2024-03-13:
Hello, I am having a similar issue even after downgrading to pyexiftool==0.4.13.

I got he following error when trying the Setup.ipynb  : `---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[2], line 5
      2 imagePath = os.path.join('.','data','REDEDGE-MX')
      3 imageName = glob.glob(os.path.join(imagePath,'IMG_0001_1.tif'))[0]
----> 5 img = Image(imageName)
      6 img.plot_raw(figsize=(8.73,8.73));
      8 from micasense.panel import Panel

File ~\imageprocessing\micasense\image.py:72, in Image.__init__(self, image_path, exiftool_obj, allow_uncalibrated)
     70     raise IOError("Provided path is not a file: {}".format(image_path))
     71 self.path = image_path
---> 72 self.meta = metadata.Metadata(self.path, exiftool_obj=exiftool_obj)
     74 if self.meta.band_name() is None:
     75     raise ValueError("Provided file path does not have a band name: {}".format(image_path))

File ~\imageprocessing\micasense\metadata.py:51, in Metadata.__init__(self, filename, exiftool_path, exiftool_obj)
     49 if not os.path.isfile(filename):
     50     raise IOError("Input path is not a file")
---> 51 with exiftool.ExifToolHelper() as exift:
     52     self.exif = exift.get_metadata(filename)

AttributeError: module 'exiftool' has no attribute 'ExifToolHelper'
`
Is there a solution for this? Any help, please (a few days trying to figure out)

AdrienWehrle, 2024-06-28:
> > Hello, I am having a similar issue even after downgrading to pyexiftool==0.4.13.
> > I got he following error when trying to run the Setup.ipynb :
> > `--------------------------------------------------------------------------- AttributeError Traceback (most recent call last) Cell In[2], line 5 2 imagePath = os.path.join('.','data','REDEDGE-MX') 3 imageName = glob.glob(os.path.join(imagePath,'IMG_0001_1.tif'))[0] ----> 5 img = Image(imageName) 6 img.plot_raw(figsize=(8.73,8.73)); 8 from micasense.panel import Panel
> > File ~\imageprocessing\micasense\image.py:72, in Image.**init**(self, image_path, exiftool_obj, allow_uncalibrated) 70 raise IOError("Provided path is not a file: {}".format(image_path)) 71 self.path = image_path ---> 72 self.meta = metadata.Metadata(self.path, exiftool_obj=exiftool_obj) 74 if self.meta.band_name() is None: 75 raise ValueError("Provided file path does not have a band name: {}".format(image_path))
> > File ~\imageprocessing\micasense\metadata.py:51, in Metadata.**init**(self, filename, exiftool_path, exiftool_obj) 49 if not os.path.isfile(filename): 50 raise IOError("Input path is not a file") ---> 51 with exiftool.ExifToolHelper() as exift: 52 self.exif = exift.get_metadata(filename)
> > AttributeError: module 'exiftool' has no attribute 'ExifToolHelper' ` Is there a solution for this? Any help, please (a few days trying to figure out)
> 
> i am also getting the same error.can anyone help me out?

Same here after downgrading to `pyexiftool==0.4.13` following the initial issue.

cgomes5, 2024-06-28:
@AdrienWehrle @Mithoon278 This usually happens if exiftool is not downloaded or not in your system's PATH. if you haven't downloaded it then use this link: https://exiftool.org/

Then add file to the systems path (instructions for windows):

Right-click on This PC or My Computer on the desktop or in File Explorer, and select Properties.
Click on Advanced system settings.
Click on Environment Variables.
In the System variables section, find the Path variable, and click Edit.
Click New, and add the path to the folder where you extracted exiftool (e.g., C:\exiftool).

Also don't forget to rename exiftool(-k).exe to exiftool.exe.

eleocharis, 2024-08-08:
I changed back:
```
exiftool.ExifToolHelper
```
to
```
exiftool.ExifTool
```

which seam to be correct again, however tah I get an similar error like above:

```micasense/test_packeges_load_tif.py:None (micasense/test_packeges_load_tif.py)
test_packeges_load_tif.py:36: in <module>
    img = Image(imageName)
image.py:72: in __init__
    self.meta = metadata.Metadata(self.path, exiftool_obj=exiftool_obj)
metadata.py:51: in __init__
    with exiftool.ExifTool() as exift:
C:\Users\Jan.Helbach\.conda\envs\micasense\lib\site-packages\exiftool\exiftool.py:301: in __init__
    raise FileNotFoundError( '"{}" is not found, on path or as absolute path'.format(self.executable) )
E   FileNotFoundError: "exiftool.exe" is not found, on path or as absolute path
```
Somebody has an Idea?

ype3, 2024-08-15:
`exiftool.ExifToolHelper(executable=os.environ.get('exiftoolpath'))`