micasense/imageprocessing #179: Tutorial 2 issue: LibRawFileUnsupportedError: b'Unsupported file format or not RAW file'-need help, please

YY-9999, 2022-03-07:
Hi everyone,

I am new to coding and micasense. I am having trouble and I just could not figure it out. Please help. Many thanks!

I followed start-up and Tutorial 1 step by step, using my own data from Micasense  RedEdge-MX ™ dual multispectral sensors. They worked perfectly. However, when I moved to Tutorial 2, I faced b'Unsupported file format or not RAW file' this problem.

The only thing I changed is image path, see as below. My own path does not have any problem, because I used same path for tutorial 1 and it worked perfectly. I just do not get it. I look forward to seeing some hints from you all. Thanks 

import os
import micasense.image as image
%matplotlib inline

image_path = os.path.join('my ownpath')
img = image.Image(image_path)
img.plot_raw();

I tried path for default 10 bands data in the folder and it worked, but it does not work for my 10 bands data. Thanks

volpatoo, 2022-03-07:
Hi. Just download the images one by one on the Github page instead of clone as recommended in Tutorial 1. For some reason, those images can not be downloaded cloning or downloaded directly. To confirm this, try to open the image on your computer, if you are able to see the image that means the file was downloaded correctly.

poynting, 2022-03-07:
See pinned issue about git-lfs. This is likely the reason.

YY-9999, 2022-03-08:
Hi @volpatoo and @poynting 

Thank you both for helping.

I can open all my own images in my computer, which means image file was downloaded correctly.

I just run the single image into codes below again, it still does not work and has a same issue.

default codes using default image, it worked:

import os
import micasense.image as image
%matplotlib inline
image_path = os.path.join('.','data','0000SET','000','IMG_0000_1.tif')
img = image.Image(image_path)
img.plot_raw();

default codes using my own image, I moved my image into same folder and changed name into 'IMG_0000_6.tif'. It did not work:

import os
import micasense.image as image
%matplotlib inline
image_path = os.path.join('.','data','0000SET','000','IMG_0000_6.tif')
img = image.Image(image_path)
img.plot_raw();

I also reinstall git and git-lfs. Unfortunately, it still does not work. 

I tried so many different way, change image name and path, compared my image with default image, googled the issues and wrote new codes. No solutions come up with :(

Any idea? Thanks in advance.

YY-99999, 2022-05-12:
Hi all,

just a quick update in case someone is having same issues as mine. The issue is raw file is not 16-bit. All codes are suitable for 16-bit TIFF to 12-bit DNG. Thanks

closed 2022-05-12 by YY-99999