micasense/imageprocessing #182: How to convert the raw data into pixel-scale irradiance?

zhuqingqi, 2022-04-07:
Hi
I don't know whether I could obtain the pixel-scale irradiance of ground surfaces. I have tried "batch processing example", however, the results seem to be radiance instead of irradiance. Is it possible for me to multiply the pixel-scale raidance by math.pi for getting the irradiance of each pixel?

*Please notice that I focus on obtaining the irradiance of each pixel based on raw data of Rededge-MX. More details are explained in the Part 2.6 of this paper. DOI: [10.3390/s21144847](https://doi.org/10.3390/s21144847)

Moreover, the pixel of stacks generated from the "batch processing example" seems to be gray values (picture attached) instead of radiance. How do I convert these gray values into radiance?

[image]

[image]

Thanks for your reading

fdarvas, 2022-04-26:
The Image.undistorted_radiance() method does return a floating point numpy array of the undistorted radiance. This can be multiplied by pi to get an estimate of the irradiance.

zhuqingqi, 2022-05-15:
@fdarvas Thanks. However, the pixel of results generated from the "batch processing example" seems to be DN values or RGB (picture attached) instead of radiance. How do I convert these DN values/RGB into radiance? Thanks
[image]
[image]
[image]

Hope to hear from you. Thanks

poynting, 2022-09-15:
To do this you'll need to know the reflectance of the surface, and assume it's lambertian. We did that [here](https://micasense.github.io/imageprocessing/MicaSense%20Image%20Processing%20Tutorial%203.html#Tying-the-camera-and-DLS-together-via-Panels) in one of the tutorials to estimate the irradiance of the panel knowing it's radiance and reflectance (albedo).