micasense/imageprocessing #183: MicaSense ® RedEdge Multi-channel Image Registration

wanghanruibaba, 2022-04-11:
I am using the MicaSense ® RedEdge camera to take close-up pictures. Due to the various imaging systems in each channel, a phase difference is generated in the captured 5-channel images. How to align and register the 5-channel images, looking forward to your reply ,thanks

fdarvas, 2022-04-26:
For shots at close distances, a band-to-band alignment of different channels - such as e.g. red to green - requires the solution of a dense stereo matching problem.

A method for this is described here:

High-density stereo image matching using intrinsic curves
December 2018[ISPRS Journal of Photogrammetry and Remote Sensing](https://www.researchgate.net/journal/ISPRS-Journal-of-Photogrammetry-and-Remote-Sensing-0924-2716) 146:373-388
DOI: [10.1016/j.isprsjprs.2018.10.005](http://dx.doi.org/10.1016/j.isprsjprs.2018.10.005)

Ideally this would be done between two channels that are most similar in the spectral response (such as green and blue) and are the most separated.

If a solution for the dense matching is found, i.e. basically a full depth map for each pixel in e.g. the green channel is available, a mapping to the other bands can be found via construction of the trifocal tensor. 
At this point, only the camera rig-relatives and translations are required, which can be found in the image metadata.

This is described in detail here:

Shahbazi, M., and C. Cortes. "Seamless Co-Registration of Images from Multi-Sensor Multispectral Cameras." The International Archives of Photogrammetry, Remote Sensing and Spatial Information Sciences 42 (2019): 315-322.
https://pdfs.semanticscholar.org/dff5/e50ebed805f18ca03ec9e95fdb5a4c507801.pdf

This is a rather complex problem and thus we will not support it in this toolbox.

wanghanruibaba, 2022-04-27:
Your reply gave me ideas, thank you. However, the article High-density stereo image matching using intrinsic curves does not have download permission, it would be great if you can provide the original text. Thanks again

SuperDiodo, 2024-11-25:
@wanghanruibaba , have you succeeded in implementing the solution proposed by @fdarvas? I'm also interested in aligning captures with frontal/close targets.