micasense/imageprocessing #187: Revisiting a closed issue

Christine1427, 2022-06-24:
Hi, I have this same issue. Working from a new environment as well which I cloned the micasense imageprocessing into. When I then try to      import micasense.metadata as metadata      the error produced is ModuleNotFoundError: No module named 'exiftool'. I have also triple checked I downloaded exiftool properly and is located right on the C drive. 
Thank you for your time, 
Christine

_Originally posted by @Christine1427 in https://github.com/micasense/imageprocessing/issues/134#issuecomment-1165778998_

wanghanruibaba, 2022-09-11:
In system variables or my variables, add the environment variable path for exiftool

poynting, 2022-09-15:
There is the exiftool executable, which you'll need to add the path and environment variable for. Then there's the pyexiftool python module that's installed via pip when the environment is created.  If something went wrong you might not have the module, which would create this error.  You can try to install it via pip with `pip install pyexiftool<=0.4.13`