micasense/imageprocessing #189: Reading RedEdge Metadata¶

wanghanruibaba, 2022-07-20:
There was an error reading metadata using the code in the tutorial.

import micasense.metadata as metadata
import os

imagePath = os.path.join('.','data','0000SET','000')
imageName = os.path.join(imagePath,'IMG_0000_4.tif')
exiftoolPath = None
if os.name == 'nt':
    exiftoolPath = os.environ.get('exiftoolpath')
# get image metadata
meta = metadata.Metadata(imageName, exiftoolPath=exiftoolPath)
cameraMake = meta.get_item('EXIF:Make')
cameraModel = meta.get_item('EXIF:Model')
firmwareVersion = meta.get_item('EXIF:Software')
bandName = meta.get_item('XMP:BandName')
print('{0} {1} firmware version: {2}'.format(cameraMake,
                                             cameraModel,
                                             firmwareVersion))
print('Exposure Time: {0} seconds'.format(meta.get_item('EXIF:ExposureTime')))
print('Imager Gain: {0}'.format(meta.get_item('EXIF:ISOSpeed')/100.0))
print('Size: {0}x{1} pixels'.format(meta.get_item('EXIF:ImageWidth'),meta.get_item('EXIF:ImageHeight')))
print('Band Name: {0}'.format(bandName))
print('Center Wavelength: {0} nm'.format(meta.get_item('XMP:CentralWavelength')))
print('Bandwidth: {0} nm'.format(meta.get_item('XMP:WavelengthFWHM')))
print('Capture ID: {0}'.format(meta.get_item('XMP:CaptureId')))
print('Flight ID: {0}'.format(meta.get_item('XMP:FlightId')))
print('Focal Length: {0}'.format(meta.get_item('XMP:FocalLength')))

Traceback (most recent call last):
  File "D:/Pycharm Projiect/Macisense test/micasense/test7.py", line 10, in <module>
    meta = metadata.Metadata(imageName, exiftoolPath=exiftoolPath)
  File "D:\Pycharm Projiect\Macisense test\micasense\metadata.py", line 47, in __init__
    raise IOError("Input path is not a file")
OSError: Input path is not a file

imageName is a relative address, not a file

poynting, 2022-09-15:
https://github.com/micasense/imageprocessing/issues/183

wanghanruibaba, 2022-09-16:
Thank you for your reply, but, using micasense at close range, the DN value is converted into reflectance, is there a corresponding tutorial? Express my gratitude to you, thank you

poynting, 2022-09-16:
Converting DN to reflectance from close range is no different than from a distance.  The challenge with close range is that the lighting can be uneven due to reflections from close by objects, or, in the case of inside a greenhouse, the lighting is very inconsistent. The other challenge is that due to parallax effects, the images are difficult to align without substantially different methods. This is why you were provided that answer by @fdarvas in your other issue https://github.com/micasense/imageprocessing/issues/183#issuecomment-1110277989

[Please start with the tutorial articles](https://micasense.github.io/imageprocessing/index.html)

wanghanruibaba, 2022-09-16:
Thank you for your patience in replying and expressing gratitude to you