micasense/imageprocessing #191: Alignment of Bands in RedEdge-P

MarioCruz22, 2022-09-16:
Getting in trouble with metadata differences like Vignette and Radiance and so on... Anyone knows how to fit RedEdge-P into the tutorial available at https://micasense.github.io/imageprocessing/Alignment.html? Or any other tutorial that can teach me how to treat RedEdge-P data and align its bands?

RedEdge
Vignetting Center : 639.29433505013424, 480.39791730098648
Vignetting Polynomial : -0.00035899158967688416, 3.8849091850333786e-06, -2.057088751909051e-08, 5.0152576116649375e-11, -5.827880227440714e-14, 2.5353631877162346e-17

RedEdge-P
Vignetting Polynomial 2D Name : 0,0,0,1,0,2,0,3,0,4,0,5,1,0,1,1,1,2,1,3,1,4,2,0,2,1,2,2,2,3,3,0,3,1,3,2,4,0,4,1,5,0
Vignetting Polynomial 2D : 0.467640,0.501446,-0.288309,-0.448964,0.306178,-0.083137,1.384667,0.284610,-0.472609,0.215251,-0.015028,-0.951077,-0.142211,0.523393,-0.179298,-0.884722,-0.275812,-0.063004,0.511571,0.124509,-0.058204

MarioCruz22, 2022-09-22:
I was wondering if are there any support that helps me to create a vignetting correction matrix with the metadata provided by RedEdge-P sensors!

Vignetting Polynomial 2D Name : 0,0,0,1,0,2,0,3,0,4,0,5,1,0,1,1,1,2,1,3,1,4,2,0,2,1,2,2,2,3,3,0,3,1,3,2,4,0,4,1,5,0
Vignetting Polynomial 2D : 0.467640,0.501446,-0.288309,-0.448964,0.306178,-0.083137,1.384667,0.284610,-0.472609,0.215251,-0.015028,-0.951077,-0.142211,0.523393,-0.179298,-0.884722,-0.275812,-0.063004,0.511571,0.124509,-0.058204

andrematte, 2023-01-02:
Hi @MarioCruz22 , I'm facing the same issues when trying to process images taken with a RedEdge-P using open-source solutions. Did you find any solution or materials?

MarioCruz22, 2023-01-03:
Yup! I did find a solution but I do not recall where.. u can take a loot to "https://forum.developer.parrot.com/t/vignetting-correction-sample-code/5614", as it seems close to what I have, however my results didn't seem quite well nor accurate with corrections. After all I just dropped vignetting corrections, for now.

MarioCruz22, 2023-01-03:
Not close to my laptop right now, but as soon as I get my code I'll post it here.

andrematte, 2023-01-03:
Thank you very much, that would be really helpful!

HazelSCUT, 2023-08-23:
Hi, I have also encountered such a problem. The data cannot be compatible with Rededge-P. How could you solve such a problem? It seems that it is impossible to open the Rededge-P images, not to mention later processing (e.g., alignment or batch processing):
[image]
[image]
Looking forward to your reply. Many thanks.

alherca73, 2023-10-24:
Hello @MarioCruz22 , did you modify your scripts so that imagery from RedEdge-P or Altum-PT can be processed from DN to Reflectance values?

MarioCruz22, 2023-12-12:
I did not yet… Did you? I am now back to work with this library and will try to fix these incompatibilities. Have some or any of you already managed to get reflectance values from DN or the 'vignette_polynomial_2D_name' from 'Vignetting Polynomial'?
I also noticed that one link that I have commented above is now broke. Sorry for that!

closed 2024-03-02 by MarioCruz22