micasense/imageprocessing #193: All .tif files truncated to 132 bytes

tomoreilly, 2022-10-21:
I cloned the project, and all of the .tif files under data/ are only 132 bytes.
I also downloaded the master branch .zip files, and those files are also 132 bytes.

tomoreilly, 2022-10-21:
I did some searching around - I installed git-lfs and now the full-size tif files are cloned.

closed 2022-10-21 by tomoreilly

maneaba, 2022-10-21:
You need to install and enable Git LFS before installation. Try doing this,
then trying to clone the project again.

tomoreilly, 2022-10-21:
Thanks, I installed/enabled git-lfs and that successfully clones the full .tif files. 
I closed the issue at github. 

Thanks 
Tom 

-------------------------------------------------- 
Thomas C. O'Reilly 
Monterey Bay Aquarium Research Institute 
7700 Sandholdt Road 
Moss Landing, California 95039-9644 
831-775-1766 (voice) 
831-775-1620 (FAX) 
***@***.*** (email) 
http://www.mbari.org (World-wide Web) 

"The machine does not isolate us from the great mysteries 
of nature, but plunges us more deeply into them." 

- ANTOINE DE SAINT-EXUPERY 
"Wind, Sand, and Stars" (1939) 

Stephen