micasense/imageprocessing #198: Many codes are uncompatible with RedEdge-p

HazelSCUT, 2023-08-18:
I tried MicaSense Image Processing Tutorial 2,3 to process images derived from RedEdge-p. However, it seems that such packages cannot work:
[image]
[image]
[image]
How to solve such problems? Should the library be updated according to RedEdge-p? Many thanks.

HazelSCUT, 2023-08-18:
Additionally, when I use micasense.Capture, I got the same error whether I tried images deroved from Rededge-MX or Rededge-P:
[image]
[image]
[image]

alherca73, 2023-10-24:
Hello @HazelSCUT , have you found a way to work with either RedEdge-P or Altum-PT using these scripts?