micasense/imageprocessing #199: Stuck at Solving Environment on Anaconda

timothy-krock, 2023-08-30:
Hey I ran into an issue while working through [MicaSense Image Processing Setup](https://micasense.github.io/imageprocessing/MicaSense%20Image%20Processing%20Setup.html#MicaSense-Image-Processing-Setup) on my windows machine where `conda env create -f micasense_conda_env.yml` would hang on "Solving environment" for hours. 

Forcing conda forge fixed it for me: 
`conda config --set channel_priority strict`

Might be good to include in the setup. 

[This stack overflow provided the fix](https://stackoverflow.com/questions/63734508/stuck-at-solving-environment-on-anaconda)

snehitvaddi, 2023-10-08:
Same problem here while installing on local system.
 I attempted to set it up on Google Colab, assuming it would simplify the process, but I'm still facing challenges. It's quite frustrating to consistently see the "exiftool" errors agin and again.🫠

sebc06, 2024-02-05:
@timothy-krock please update to the latest version of Conda or force the usage of samba solver

conda update conda
conda install -n base conda-libmamba-solver
conda config --set solver libmamba

Note that this is now the default solver since Conda 23.10.0.

sebc06, 2024-02-06:
@snehitvaddi exiftools issues were fixed recently. Please update your local repository. It the issue persists, open a separate issue.

sebc06, 2024-02-26:
Closing this issue.

closed 2024-02-26 by sebc06