micasense/imageprocessing #201: Will there be support for the newer sensors RedEdge-P and Altum-PT?

alherca73, 2023-10-24:
Hello all! - I think the scripts in this repository provide a lot of great guidance. Unfortunately, there is not support for the newer sensors RedEdge-P and Altum-PT. I can see that a lot of issues that deal with these two newer sensors are still open and they have never been addressed by @poynting - is there any plans for the near future to modify these scripts so that RedEdge-P and Altum-PT are supported? or is the idea that people should depend on commercial software to process the imagery from these two sensors? Thanks for any guidance, or advise that can be provided.

hjbigloo, 2023-12-22:
This is absurd. We got this camera (Altum PT) to use it in scientific research and after collecting data for a year, now nothing works. Apparently, they have changed some exif parameters in the images and the old code is not working with the new images. In particular they have replaced 'XMP:VignettingCenter' with vignetting polynomial and they have not released any article or script explaining how to use this. We have trying to solve this issue for more than two months now and there is no proper support. They have updated the repository last week, but still it is not working. 
Do not waste your money on this garbage.

sebc06, 2024-02-29:
Support for Panchro cameras added. Closing this issue.

closed 2024-02-29 by sebc06