micasense/imageprocessing #203: ValueError: Provided file path does not have a band name

Pepe-oss, 2023-11-05:
Hi.
When I finished running `conda env create -f micasense_conda_env.yml` . When I run the subsequent test code for "MicaSense Image Processing Setup.ipynb", I can successfully run the first stack and all the packages needed for testing are loaded. But for the second stack, I get the following error when I run it:
`ValueError: Provided file path does not have a band name: .\data\0000SET\000\IMG_0000_1.tif`
Has anyone else encountered this problem, also, I used qgis to go and open that tif file directly and it won't open.

fuxinyang0708, 2023-11-06:
Maybe your tif files are broken. I think you can redownload the tif files  from github.

Pepe-oss, 2023-11-06:
> Maybe your tif files are broken. I think you can redownload the tif files from github.

OK, you are right. Thanks!

closed 2023-11-06 by Pepe-oss