micasense/imageprocessing #208: Converting .ipynb to .py issue: display()

tomoreilly, 2024-02-01:
I am converting Alignment-10Band.ipynb to a straight .py script. 
The ipynb page invokes display() as follows:

```
im_aligned = imageutils.aligned_capture(capture, warp_matrices, warp_mode, cropp
ed_dimensions, match_index, img_type=img_type)
display(im_aligned)
```
But the .py script is unable to find symbol display:

>   File "Alignment-10Band.py", line 189, in main
>     display(im_aligned)
> NameError: name 'display' is not defined
> 
What should my .py script call instead of display(im_aligned)?
Thanks!

sebc06, 2024-02-05:
Hi @tomoreilly,
This line should be removed. I'll merge the change soon. You can remove locally in the meantime.

sebc06, 2024-02-09:
This is done, closing this issue.

closed 2024-02-09 by sebc06