micasense/imageprocessing #215: AttributeError: module 'exiftool' has no attribute 'ExifToolHelper'

Nldo, 2024-03-25:
Hello, I encountered an error while running the code: AttributeError: module 'exiftool' has no attribute 'ExifToolHelper'. I tried using pyexiftool versions 0.4.13 and 0.5.6, but it didn't work

Nldo, 2024-03-25:
Setting the Python environment to 3.10 can solve the problem.

hanhern, 2024-07-08:
Has anyone else had this issue? I get the problem using python 3.8 and python 3.10. When I print(dir(exiftool)), I get the following attributes: '__class__', '__del__', '__delattr__', '__dict__', '__dir__', '__doc__', '__enter__', '__eq__', '__exit__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_check_sanity_of_result', 'copy_tags', 'execute', 'execute_json', 'execute_json_wrapper', 'get_metadata', 'get_metadata_batch', 'get_metadata_batch_wrapper', 'get_metadata_wrapper', 'get_tag', 'get_tag_batch', 'get_tag_batch_wrapper', 'get_tag_wrapper', 'get_tags', 'get_tags_batch', 'get_tags_batch_wrapper', 'get_tags_wrapper', 'set_keywords', 'set_keywords_batch', 'set_tags', 'set_tags_batch', 'start', 'terminate'

chengjian-zhang, 2024-07-23:
> Setting the Python environment to 3.10 can solve the problem.

I have set python to 3.10.0, but it still reports the same error.